malachi-iot/estdlib #1: Using estdlib in platformio

seanngpack, 2020-01-01:
Hi, I'd like to use your library in platformIO, more specifically utilize the cstddef header bundled in the library. Are there any examples on how to use this in platformIO?

malachib, 2020-01-05:
@seanngpack Hi, technically a partial yes with:

https://github.com/malachi-iot/estdlib/tree/dev/test/arduino/basic

That hasn't been tested in a while however.  If that doesn't get you going hit me with your specifics and we'll come up with a solution

malachib, 2020-01-05:
Hi @seanngpack 

Give https://github.com/malachi-iot/estdlib/tree/dev/examples/pio/string a try see how that works out for you

malachib, 2020-01-20:
Made some revisions, this one is a better fit for you:

https://github.com/malachi-iot/estdlib/blob/master/test/unity/cstddef.cpp

What environment/MCU are you using?

closed 2020-02-07 by malachib