malachi-iot/estdlib #3: cannot use estdlib with basic uno

justin-romano, 2020-04-21:
just instantly hopeless. Any ideas?

platformio.ini
lib_deps =
    estdlib@0.1.4
    JSON

```console
> Executing task: C:\Users\Administrator\.platformio\penv\Scripts\platformio.exe run <

Processing uno (platform: atmelavr; board: uno; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
PACKAGES:
 - framework-arduino-avr 5.0.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)      
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <estdlib> 0.1.4
|-- <EEPROM> 2.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\uno\src\main.cpp.o
In file included from .pio\libdeps\uno\estdlib_ID5784\src/estd/type_traits.h:12:0,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/internal/../port/arduino/Print.h:8,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/internal/../port/support_platform.h:11,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/internal/platform.h:189,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/internal/type_traits.h:9,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/utility.h:9,
                 from .pio\libdeps\uno\estdlib_ID5784\src/estd/map.h:3,
                 from src\main.cpp:7:
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:9:33: error: expected class-name before '{' token
 struct is_function : false_type { };
                                 ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:13:46: error: expected class-name before '{' token
 struct is_function<Ret(Args...)> : true_type {};
                                              ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:17:52: error: expected class-name before '{' token
 struct is_function<Ret(Args...) const> : true_type {};
                                                    ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:19:55: error: expected class-name before '{' token
 struct is_function<Ret(Args...) volatile> : true_type {};
                                                       ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:21:61: error: expected class-name before '{' token
 struct is_function<Ret(Args...) const volatile> : true_type {};
                                                             ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:37:48: error: expected class-name before '{' token
 struct is_function<Ret(Args...) &> : true_type {};
                                                ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:39:54: error: expected class-name before '{' token
 struct is_function<Ret(Args...) const &> : true_type {};
                                                      ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:41:57: error: expected class-name before '{' token
 struct is_function<Ret(Args...) volatile &> : true_type {};
                                                         ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:43:63: error: expected class-name before '{' token
 struct is_function<Ret(Args...) const volatile &> : true_type {};
                                                               ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:55:49: error: expected class-name before '{' token
 struct is_function<Ret(Args...) &&> : true_type {};
                                                 ^
.pio\libdeps\uno\estdlib_ID5784\src/estd/internal/is_function.h:57:55: error: expected class-name before '{' token
 struct is_function<Ret(Args...) const &&> : true_type {};
[... 521 lines cut ...]
.pio\libdeps\uno\estdlib_ID5784\src/estd/map.h: In constructor 'estd::map<Key, T, Compare, Allocator>::map(estd::map<Key, T, Compare, Allocator>::initializer_list, bool, const Compare&, const allocator_type&)':
.pio\libdeps\uno\estdlib_ID5784\src/estd/map.h:107:9: error: 'typedef int estd::map<Key, T, Compare, Allocator>::base_t' is not a non-static data member of 'estd::map<Key, T, Compare, Allocator>'    
         base_t(init, comp) {}
         ^
src\main.cpp: In member function 'String SerialRequest::GetParam(String)':
src\main.cpp:203:5: warning: no return statement in function returning non-void [-Wreturn-type]
     }
     ^
src\main.cpp: In function 'void loop()':
src\main.cpp:279:45: warning: large integer implicitly truncated to unsigned type [-Woverflow]
                     delayMicroseconds(100000);
                                             ^
src\main.cpp:287:39: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                     for (int i = 0; i < ARRAYSIZE(analogWaterSensorPorts); i++)
                                       ^
src\main.cpp: At global scope:
src\main.cpp:45:1: warning: '<anonymous>' defined but not used [-Wunused-variable]
 };
 ^
*** [.pio\build\uno\src\main.cpp.o] Error 1
===================================================================================== [FAILED] Took 3.59 seconds =====================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

malachib, 2020-05-08:
Hm!  Did not expect that.  I will have a look, thank you for bringing this to my attention

malachib, 2020-05-11:
Laest `alpha` branch has a fix for this.  If you are able to test directly against this github branch, please do so.  Otherwise I'll publish 0.1.6 as time permits

justin-romano, 2020-05-12:
Thanks. hope it was not a biggy. Must be a tough port from stl to avr.

malachib, 2020-05-23:
In the past this was actually tested against AVR, it's more a C++ thing.  I think the 0.1.6 prerelease will work for you.  Let me know!

malachib, 2020-05-30:
> Thanks. hope it was not a biggy. Must be a tough port from stl to avr. Especially with the memory constraints

I didn't understand your comment completely the first time around -- the port to AVR wasn't too terrible, everything is a ground up re-write and that actually made it easier

closed 2020-06-04 by malachib