malachi-iot/estdlib #4: missing features.h

HotelCalifornia, 2020-09-07:
estd/port/gnuc.h:17-27

```cpp
#if defined(__linux__) || defined(__GLIBC__)
// Only a linux/glibc thing:
// https://stackoverflow.com/questions/259248/how-to-test-the-current-version-of-gcc-at-compile-time
// GLIBC detection gleaned from
// http://web.mit.edu/freebsd/head/contrib/wpa/src/utils/common.h
#include <features.h>
#else
#ifndef __GNUC_PREREQ
#define __GNUC_PREREQ(major, minor) (__GNUC__ > major || (__GNUC__ == major && __GNUC__MINOR__ >= minor))
#endif
#endif
```

<details>
<summary>for reference, here's (one of) the one(s) used by the arm-none-eabi toolchain</summary>

```cpp
/*
 *  Written by Joel Sherrill <joel@OARcorp.com>.
 *
 *  COPYRIGHT (c) 1989-2014.
 *
 *  On-Line Applications Research Corporation (OAR).
 *
 *  Permission to use, copy, modify, and distribute this software for any
 *  purpose without fee is hereby granted, provided that this entire notice
 *  is included in all copies of any software which is or includes a copy
 *  or modification of this software.
 *
 *  THIS SOFTWARE IS BEING PROVIDED "AS IS", WITHOUT ANY EXPRESS OR IMPLIED
 *  WARRANTY.  IN PARTICULAR,  THE AUTHOR MAKES NO REPRESENTATION
 *  OR WARRANTY OF ANY KIND CONCERNING THE MERCHANTABILITY OF THIS
 *  SOFTWARE OR ITS FITNESS FOR ANY PARTICULAR PURPOSE.
 *
 *  $Id$
 */

#ifndef _SYS_FEATURES_H
#define _SYS_FEATURES_H

#ifdef __cplusplus
extern "C" {
#endif

#include <_newlib_version.h>

/* Macro to test version of GCC.  Returns 0 for non-GCC or too old GCC. */
#ifndef __GNUC_PREREQ
# if defined __GNUC__ && defined __GNUC_MINOR__
#  define __GNUC_PREREQ(maj, min) \
	((__GNUC__ << 16) + __GNUC_MINOR__ >= ((maj) << 16) + (min))
# else
#  define __GNUC_PREREQ(maj, min) 0
# endif
#endif /* __GNUC_PREREQ */
/* Version with trailing underscores for BSD compatibility. */
#define	__GNUC_PREREQ__(ma, mi)	__GNUC_PREREQ(ma, mi)

/*
 * Feature test macros control which symbols are exposed by the system
 * headers.  Any of these must be defined before including any headers.
 *
 * __STRICT_ANSI__ (defined by gcc -ansi, -std=c90, -std=c99, or -std=c11)
 *	ISO C
 *
 * _POSIX_SOURCE (deprecated by _POSIX_C_SOURCE=1)
 * _POSIX_C_SOURCE >= 1
[... 479 lines cut ...]

#ifdef __cplusplus
}
#endif
#endif /* _SYS_FEATURES_H */
```
</details>

Just skimming through estdlib, I don't think you use any feature test macros (I imagine a lot of that is just done by switching platform includes) so I tried just adding the minimal

```cpp
#ifndef _FEATURES_H
#define _FEATURES_H

#ifdef __cplusplus
extern "C" {
#endif

#ifdef __cplusplus
}
#endif
#endif /* _FEATURES_H */
```

and the compiler quit complaining.

malachib, 2020-09-09:
I use a few compilers that don't have those feature tests available.  But I concur it is the proper way :)  Truthfully, I implemented it before I knew those existed...

Odd that features.h itself is giving you troubles, I'll have a look

malachib, 2020-09-09:
OK it seems that GLIBC availability is somewhat tricky.  <features.h> is generally expected if GLIBC is present, but something like newlib isn't _exactly_ GLIBC just very very close so <features.h> isn't available in that context.  I assume you're using an ARM32 environment with no glibc-features or newlib?  Or rather, please share your environment info including host OS and target MCU

HotelCalifornia, 2020-09-10:
I'm targeting an Arduino pro mini---I only mentioned the stuff about Arm as a point of comparison as it's the only other cross-compiler toolchain I had on hand.

Host OS is (effectively) Ubuntu 20

malachib, 2020-09-11:
The AVR is probably close enough in this context.  I happen to have a few of those.

How are you interacting with the toolchain?  Are you using Arduino IDE, platformio, raw/direct, or something else?

HotelCalifornia, 2020-09-12:
> How are you interacting with the toolchain? Are you using Arduino IDE, platformio, raw/direct, or something else?

I'm using platformio.

(sorry for my delayed responses, still not in the habit of looking at the new github notifications)

malachib, 2020-09-13:
I'm having trouble recreating your issue.  How does your project's `platformio.ini` look?  Compare to the one under `examples/pio/string`

I'm wondering if you have an onhand AVR toolchain which is taking precedence over platformio's (doubtful, but possible)

HotelCalifornia, 2020-09-13:
platformio.ini:

```ini
[env:pro16MHzatmega328]
platform = atmelavr
board = pro16MHzatmega328
framework = arduino
lib_deps = 
	adafruit/Adafruit SSD1306@^2.4.0
	malachi-iot/estdlib@^0.1.6
	adafruit/Adafruit BusIO@^1.4.2
```

which looks pretty ok compared to the example one

malachib, 2020-09-13:
I was running an earlier version of `platformio` and was getting this error with that configuration:

```
AssertionError: Package version 6de96cb doesn't satisfy requirements ^0.1.6:
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/builder/main.py", line 156:                                                                    
    env.SConscript("$BUILD_SCRIPT")                                                                                                                                
  File "/home/malachi/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 597:                                                      
    return _SConscript(self.fs, *files, **subst_kw)                                                                                                                
  File "/home/malachi/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:                                                      
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)                                                                                          
  File "/home/malachi/.platformio/platforms/atmelavr/builder/main.py", line 161:                                                                                   
    target_elf = env.BuildProgram()                                                                                                                                
  File "/home/malachi/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:                                                            
    return self.method(*nargs, **kwargs)                                                                                                                           
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 62:                                                         
    env.ProcessProjectDeps()                                                                                                                                       
  File "/home/malachi/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:                                                            
    return self.method(*nargs, **kwargs)                                                                                                                           
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/builder/tools/platformio.py", line 131:                                                        
    project_lib_builder = env.ConfigureProjectLibBuilder()                                                                                                         
  File "/home/malachi/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:                                                            
    return self.method(*nargs, **kwargs)                                                                                                                           
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 1036:                                                           
    project.install_dependencies()                                                                                                                                 
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/builder/tools/piolib.py", line 868:                                                            
    lm.install(uri)                                                                                                                                                
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/managers/lib.py", line 281:                                                                    
    force=force,                                                                                                                                                   
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/managers/package.py", line 698:                                                                
    pkg_dir = self._install_from_url(name, url, requirements, track=True)                                                                                          
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/managers/package.py", line 499:                                                                
    return self._install_from_tmp_dir(_tmp_dir, requirements)                                                                                                      
  File "/home/malachi/.local/lib/python3.7/site-packages/platformio/managers/package.py", line 542:                                                                
    assert tmp_manifest["version"] == requirements, mismatch_error
```

However, after upgrading platformio to 5.0.1, I am once again getting successful compiles

Successful compiles aside, I can see that there is a potential issue in that area you pointed out so I am looking into it.  If you have the ability to do a direct git checkout, pull from the `github-issue-4` branch and see what your mileage is there

malachib, 2022-05-30:
Stale issue, closing

closed 2022-05-30 by malachib